ftCommunity/ftcommunity-TXT #190: The future

ski7777, 2020-07-01:
# Current situation

- We are on buildroot 2018.11.2
- We use Qt 4
- Most security relevant packages are outdated
- There are some contributors left working on minor bugs and willing to improve the whole project but they are slowed down by other people

# What I think should be changed

- Update to a newer version of buildroot or another distribution
- Update to Qt5 or use something else
- Update all security packages and keep updating them
- Encourage people to make bigger contributions and if maintainers consider to stop using this project anymore they should stop slowing down other contributors

# Concrete concept

## Distribution
For now we should keep using buildroot. Currently we have only some minor customizsations  so it should be quite easy to use another distribution if needed

## Graphics library
Qt4 has had its own window manager and could access the framebuffer and touchscreen directly. This system was very lightweight.
Qt5 does not  have a window manager that has the ability to manage multiple applications at a time. If we would only have one application running we could still have Qt5 manage the framebuffer and the touchscreen. Otherwise we would need a real window manager like Xorg but that would be too much load for the TXT.
Currently app development is quite complicated and does not allow to use modern html/js-webapps. I would approach both the window manager and the html problem by just running a simple web browser on the TXT. There will be a secure API to make system calls for all applications. I don't want to go into detail here now. If there are any design questions regarding this topic we can discuss that in another issue.

## Security-Relevant packages
We just need to update buildroot as fast as possible and create an update package.

## Maintainership
I don't want to be responsible for this whole project. I just want to be someone who creates new features, fixes bugs and makes this project a benefit for every user. It would be nice if everyone who does not want to use this project anymore would just let the other people do their work. Thanks.

Please give feedback on my proposals

Raphael

heuer, 2020-07-02:
"The future" is somewhat unspecific.

What periods are you talking about?
Switching to Qt5 is probably a good point because Qt4 will no longer be supported in the foreseeable future, but I think it is more important that the cfw offers a stable, usable platform.
The existing applications will not be available overnight under Qt5 or as a web application.
Any castles in the air will not help the project, should the changeover take place in a short time, it will throw the project back rather than bring it forward.

In my opinion, a switch to Qt5 could take place relatively quickly, as far as I know the launcher can already run largely under Qt5. But there is no rush for this as long as Qt4 is still supported. The latter is, to my knowledge, still supported by all Linux distributions. Does buildroot differ?

As soon as a stable version, e.g. 1.0, has been released, you may be able to think about other approaches, currently the efforts are not helping because they take away resources from the project.

I am not so much involved in the development of cfw, but I have the impression that new, supposedly innovative ideas are constantly being interspersed, empty and undocumented repositories are created, but ultimately it does not help anyone and it definitely won't help users. A user is interested in why he cannot run App XY because he is interested in it. He is not interested in whether Qt4, Qt5, LVGL or a WebUI is used.

In my opinion, the focus should be on the user and stable, usable platform. After a final release of the current approach your point "graphic library" should be considered and there may be good reasons to reject or adjust the current approach, but at the moment, I don't think such a big change fits the concept, even with the few developers in mind.

ski7777, 2020-07-02:
A switch to qt5 on the cfw is nearly impossible. I don't think anybody is going to do that.
You are absolutely right that a transition of the apps to Qt5 would take some time and rewriting them to use html would cost even more time. Even though I think we need to create a stable system and working on a Qt5 solution is waste of time.

Yes, there are some repositories with just few/none documentation. I'm working on that. If someone could delete https://github.com/ftCommunity/pkgmanager that would be nice. The usage of https://github.com/ftCommunity/gorobointerface should be clear even though I will redo that later because there will be a new usecase for that soon. Stay tuned!

ski7777, 2020-07-02:
And yes "the futire" is very unspecific. Its as unspecific as the current situation of this project. This thread is just meant to be a discussion of the next steps to make this project working and known to the ftCommunity again

ski7777, 2020-07-27:
Does anyone want to comment on this?

hypnotoad, 2022-04-04:
In my view, it would help most to update the buildroot setup and in a later step to update libraries one-by-one. Currently, it is already complicated to compile the cfw on normal Linux distributions. Once that is working, I would volunteer to keep that up-to-date as long as my txt devices work.

I agree that switching to qt5 is not good idea (qt6 is there).

PeterDHabermehl, 2022-04-04:
It's a pretty difficult situation, actually. As you may have experienced, development of the community firmware has stopped more than two years ago. Updating parts of the recent community firmware, such as Qt, would render many, if not all, available apps useless.
On the other hand, an unified community firmware for TXT, TXT4.0 and TX-Pi would be absolutely great.
I think that first aim should be to release a "final version" of the current TXT community firmware, maintaining compatibility with the currently available apps, and then look into a cfw2.0 (which may or not be compatible with all three devices mentioned above, but provide most recent software components like Qt6 and maybe also an unified hardware abstraction layer to be able to control internal I/O HW (currently ftrobopy) and the most popular external I/O HW like the previous ft interfaces (currently libroboint) , TX-Pi HAT (currently directly via RPI GPIO) or ftDuino (currently via USB, running a dedicated IO-Server-sketch), if present)
Just my thoughts, though.

hypnotoad, 2022-05-27:
I like Peter's ideas, but I on my part have no capacity on a new CWF 2.0. I on my part am trying to update the CFW to newest buildroot which mainly means to
- switch from qt3 to qt5
- migrate buildroot options
About qt migration: I qt expert mentioned that this should be doable and also qt5 to qt6 should not pose a big problem, but let's see. My branch is here: https://github.com/hypnotoad/ftcommunity-TXT/tree/new-buildroot

hypnotoad, 2022-05-29:
I think the QT migration was basically trivial (huge kudos to QT, I would not have imagined that). My branch https://github.com/hypnotoad/ftcommunity-TXT/tree/new-buildroot works with newer Linux kernel, QT5 and other up-to-date libraries. Wifi works, touchscreen and display are ok. There are some hickups / deterministic crashes that will need to be fixed, but overall it seems very stable. I will call the new version 0.10.0 for now and provide images in my repo. I would like to invite everyone to test it and to provide patches.

hypnotoad, 2022-06-05:
@PeterDHabermehl : Qt5 is surprisingly compatible to Qt4, but there are hick-ups, of course. I think the UI framework will soon be migrated and most of the apps even do not need any migration. I would propose to split of the whole UI framework into a separate repository to go one step into your proposed direction (separating firmware and CFW UIs).

hypnotoad, 2022-06-12:
I published a first 0.10 release (https://github.com/ftCommunity/ftcommunity-TXT/releases/tag/v0.10.5) on this repository. I will created issues with the tag https://github.com/ftCommunity/ftcommunity-TXT/labels/new-buildroot so we can see what is needed to be done.

What I have learned:
If we want to join forces between the TX, the Rpi and maybe the TXT4, we need some abstraction between UI and system. Currently, that is done within the launcher.

hypnotoad, 2022-12-08:
Hi all, I just pushed 0.10.7 which I consider usable enough to become the new master. It contains qt5, python3 and an up-to-date buildroot. How is the release process? Who would like to test?

hypnotoad, 2022-12-20:
I pushed 0.10.8. I think all the complicated buildroot-related adaptations and bugs should be finished now and it is time for beautifications (fonts, colors). Please test it! https://github.com/ftCommunity/ftcommunity-TXT/releases/tag/v0.10.8

ski7777, 2022-12-20:
Thanks for all your work!
I will test it in the next few days