ftCommunity/ftcommunity-TXT #191: Can't uninstall own app

Bimi124, 2020-07-16:
# Deutsch (DE)
## Versionen
 - Firmware: 0.9.5
 - Linux: 4.14.97
 - Python: 3.7.1final0
 - ftrobopy: 1.88
 - Qt: 4.8.7
 - PyQt: 4.11.3

## Beschreibung
Ich habe ein einfaches Testprogramm geschrieben (Im hier angehängten Zip-Archif wurde lediglich der Autor entfernt
([app.zip](https://github.com/ftCommunity/ftcommunity-TXT/files/4931446/app.zip))
Nach Installation der App (Die hoffentlich nicht schadet, da wo sie ist, GitHub hat das automatisch so gemacht) auf dem TXT erschien die App nun ohne jeden Unterordner auf dem Homescreen und ließ sich (über den Web-Interface) nicht deinstallieren.

Warscheinlich liegt das an den Bindestrichen in der Kathegorie. Trotzdem ist das jetzt eine dumme Situation. Kleiner Tipp: Wie wäre es, wenn die Software die App erstmal sorgfältig prüft, bevor sie die App installiert? Das würde wenigstens zukünftig solche Situationen vermeiden.

# English (EN) - Automatic translation
## Versions
 - firmware: 0.9.5
 - Linux: 4.14.97
 - Python: 3.7.1final0
 - ftrobopy: 1.88
 - Qt: 4.8.7
 - PyQt: 4.11.3
## Description
I have written a simple test program (In the attached zip-archif only the author has been removed
([app.zip](https://github.com/ftCommunity/ftcommunity-TXT/files/4931446/app.zip))

After installing the app (which hopefully doesn't hurt where it is, GitHub did it automatically) on the TXT, the app appeared on the home screen without any subfolders and could not be uninstalled (via the web interface).

This is probably due to the hyphens in the category. Nevertheless, this is now a stupid situation. A little tip: How about if the software first checks the app carefully before installing it? That would at least avoid such situations in the future.

ski7777, 2020-07-16:
Ich bin gerade im Urlaub und bin mir tatsächlich nicht ganz sicher, ob eine
Deinstallation im Launcher möglich ist. Könntest du es über da Webinterface
versuchen? Also einfach die IP-Adresse des TXT in denen Browser eingeben.
Der Rest sollte relativ intuitiv sein.

Viel Glück
Raphael

Bimi124, 2020-07-16:
Ich dachte, ich hatte geschrieben über den Webinterface?
Und der sieht bei mir so aus:
[image]
Leer.
Die Bindestriche und den langen Namen habe ich auch schon abgeändert...
Aber so dringend ist es nun auch nicht. Genieße erstmal noch ein bisschen deinen Urlaub:wink:

ski7777, 2020-07-16:
Das ist natürlich suboptimal...
Falls du dich auskennst, kannst du dich per SSH mit dem TXT verbinden und
im Ordner Apps den zugehörigen Ordner löschen. Anschließend ein Mal neu
starten und schon sollte wieder weiß funktionieren.

Viele Grüße
Raphael

Bimi124, 2020-07-17:
Es hat geklappt!
Trotzdem ärgerlich, dass ich so weit gehen musste. Naja, man kann vielleicht versuchen, das irgendwie zu verbessern, dass ich die App überhaupt installieren konnte, jedoch betrachte ich dieses Problem als Gelöst.

closed 2020-07-17 by Bimi124