ftCommunity/ftcommunity-TXT #195: TXT - MQTT- C++

OguzEmreB, 2021-09-25:
I am trying to connect txt to Thingsboard iot. I reconfigured mosquitto.conf and added ft-txt-bridge-cloud.conf which contains thingsboard settings. Yet I dont know if I configured them right. I couldnt test the connection. I need a simple code for test. Sends a value with mqtt from txt as 'method = a, params = 1'. I tried to pick some codes from factory examples but couldnt manage to do that.
Need to know which libraries I need and which codes for publish and subscribe..
Thanks..

hypnotoad, 2022-06-12:
This is the community firmware. I think there is no mqtt support available, but I think there should. I would propose to set BR2_PACKAGE_QT5MQTT as well as BR2_PACKAGE_PYTHON_PAHO_MQTT to true

hypnotoad, 2022-11-20:
v0.10.6 will have qt5mqtt and paho mqtt

closed 2022-11-20 by hypnotoad