ftCommunity/ftcommunity-TXT #196: TXT CFW Installation issue

christoschronopoulos, 2021-12-06:
Dear all,

I tried installing the Community Firmware on my TXT controller and followed the step of the installation process as follows:

- Enabled the bootloader in the TXT via PuTTY
- Enable the Boot SD in the TXT Security settings
- Unzipped the latest CFW and copied the three files to the SD (32Gb SDHC fat32)
- Booted the TXT

However, in the booting process I get a totally white screen that stays forever. It has been in that white screen for hours and I don't know whether it has some progress or not. I never get to the screen that shows the yellow loading indicator with the "community edition" text. 

Once, white it was in that white screen, I turned off the TXT controller and mounted the SD on my computer to check if anything had changed. I saw that it had created a partition . Then I formatted the SD (after deleting that partition) and started the process again. What am I doing wrong? Your recommendations and hints are very much appreciated!
Best, Christos

hypnotoad, 2022-01-30:
I have the same issue and debugged a little more with the serial console. With my txt, the white screen happens in approx 80% of the cases and it seems that the system boots up correctly - it just shows a white screen. That can be checked with "ssh -v" (if there is any kind of reply, then it continues booting).

hypnotoad, 2022-02-20:
In my case, the system boots up correctly, just the screen is white. It seems a problem with the framebuffer driver. When googling for "ili9341  white screen", it seems a common problem.

hypnotoad, 2022-02-20:
I guess this patch here would fix the problem:

https://github.com/fischertechnik/FT-TXT/blob/master/board/FT/TXT/linux_4.1.y/linux-700-lcd-white.patch

(the old initialization code contained at least 1 wrong pLcdData call)

hypnotoad, 2022-02-20:
fix is in #197

closed 2022-04-04 by hypnotoad