ftCommunity/ftcommunity-TXT #201: network system app needs to be migrated or removed

hypnotoad, 2022-06-12:
The network system app does not work properly in 0.10. Migration seems possible with reasonable work, but I am not sure if that should be done. On the one side, most current wifi devices do not allow manually setting the network settings as the wifi access point manages that. On the other side, static IP addresses allow for additional flexibility and stability.

ski7777, 2022-12-09:
I haven't tested the app yet, but what do you mean with "does not work properly"? I guess the underlying network layer did not change too much.

hypnotoad, 2022-12-09:
I think someone who knows about this app should test it. I don't know good enough what is expected. There were some minor py2 to py3 adaptations by me such that it opens without exceptions.

hypnotoad, 2022-12-19:
If the network backend did not change (I think it didn't) during the buildroot upgrade, it is just UI fixes. The first problem is the custom textbox to enter an IP address (virtual keyboard not migrated yet).

hypnotoad, 2022-12-20:
I think this is fixed in 193c952bb78b374bca909cb4c573aa3e47df421f although I don't have good ways to test this.

closed 2022-12-20 by hypnotoad