ftCommunity/ftcommunity-TXT #205: rotating mechanism for X11 needs to be implemented

hypnotoad, 2022-12-09:
cfw 0.9 had that mechanism inside qt4, for qt5 the x server has to do that

ski7777, 2022-12-09:
If it is that easy we just need a mechanism to call it and maybe preserve the settings: https://askubuntu.com/a/109537

hypnotoad, 2022-12-10:
My first try with xrandr was not successful. I think the [fbdev driver](https://www.x.org/archive/X11R6.8.0/doc/fbdev.4.html) does not support xrandr rotation.
```
# xrandr -o left --verbose
 SZ:    Pixels          Physical       Refresh
*0    240 x 320    (  64mm x  85mm )  *0   
Current rotation - normal
Current reflection - none
Rotations possible - normal 
Reflections possible - none
Setting size to 0, rotation to left
Setting reflection on neither axis
X Error of failed request:  BadMatch (invalid parameter attributes)
  Major opcode of failed request:  139 (RANDR)
  Minor opcode of failed request:  2 (RRSetScreenConfig)
  Serial number of failed request:  14
  Current serial number in output stream:  14
```

What would work is to create 4 xorg.conf files for the [specific rotations](https://www.x.org/archive/X11R6.8.0/doc/fbdev.4.html) and to include to the right one. Unfortunately, the device and the input both have to be rotated, so it is not just one place to modify. But that seems definitely feasible.

ski7777, 2022-12-10:
I'm definetly not familiar with X11. I'm thinking about a little script that generates the X11 config at startup based on some config variables (currently rotation only).

hypnotoad, 2022-12-10:
All depends on how easy it is to patch the xorg init script. I have not done that so far. Probably it would be best if this script copies the right file to the xorg.conf. I could provide 4 config files with the right settings.

hypnotoad, 2022-12-11:
https://github.com/ftCommunity/ftcommunity-TXT/commit/1920e8e61c5da23455ca7e3dbf958e74c4ba8886

ski7777, 2022-12-12:
perfect!
Can we implement this here? https://github.com/ftCommunity/ftcommunity-TXT/blob/fa8d3293061940345948e2bd9f1ebca8bfd93335/board/fischertechnik/TXT/rootfs/opt/ftc/apps/system/display/display.py#L59
I guess changing the file here would be good: https://github.com/ftCommunity/ftcommunity-TXT/blob/fa8d3293061940345948e2bd9f1ebca8bfd93335/board/fischertechnik/TXT/rootfs/etc/init.d/S90launcher#L6

hypnotoad, 2022-12-12:
I think it is simplest to leave display.py how it is and let the xorg init file copy the correct xorg.conf.foo to the xorg.conf, as specified in /etc/default/launcher.

hypnotoad, 2022-12-20:
fixed in e51b77383938dc1e84ce5dd4607155b9950fb792

ski7777, 2022-12-20:
as we are trying to keep the buildroot repo as clean as possible I would suggest putting this as a patch in the patch dir or creating a file S39XXX

hypnotoad, 2022-12-26:
I think a patchfile should be created as another ticket as there is at least one additional modification which would need to be converted to a patch file. In addition, it is not clear to me if the patchfile solution is really better (let's discuss first if it is worth the effort).

ski7777, 2022-12-26:
As I found out a the patchfile solution is impossible. buildroot patches only apply on package source code and not on buildroot source code. The init.d file comes from buildroot itself. Easiest solution:

- Delete the config file via post-build script
- Add new file (new file name #210 #213 )

ski7777, 2022-12-26:
Technically speaking we could implement some kind of an global patch aplly mechanism for the rootfs. Now by using per-package-build the rootfs is regenerated after every build thus making it possible to always apply patches

hypnotoad, 2022-12-26:
> as we are trying to keep the buildroot repo as clean as possible I would suggest putting this as a patch in the patch dir or creating a file S39XXX

To me, this seems far cleaner than scripting patches because the modification stays linked to the origin. For instance, maintaining is easier because there would be conflicts if buildroot would change filenames or similar, while otherwise some post-install script would fail and it would be hard to track which buildroot modification broke the patch.

There is also another patch already to buildroot so we have to maintain patchs to buildroot anyways.

ski7777, 2022-12-26:
Actually my idea is to get rid of the current patch situation by building the initramsfs separately

ski7777, 2022-12-26:
see #218

hypnotoad, 2022-12-27:
cleanup done in d2e030c5e5. buildroot commit is now as in master again.

closed 2022-12-27 by hypnotoad