ftCommunity/ftcommunity-TXT #210: boot animation

ski7777, 2022-12-25:
The boot logo hides at 50% and the screen goes black. Then the bar continues to around 70% without showing the logo. This does not look good and is probably caused by a wrong startup sequence. I will check on that

hypnotoad, 2022-12-25:
The reason is that the progress bar interfers with the X server. One could start the X server later after the progress bar is at 100%.

ski7777, 2022-12-25:
perfect solution but keep in mind to start the launcher later too. Alternate idea:
How early can we start the XServer? Is there a way to make this animation thing completely via X?

hypnotoad, 2022-12-26:
Currently, we use psplash. It does not seem to interfere too badly with the X server in the current configuration, so I would just draw the splash image again with xsetroot.

Similar solutions have been proposed in other places:
- https://raspberrypi.stackexchange.com/questions/75147/how-to-start-xorg-while-showing-splashscreen
- https://lists.x.org/archives/xorg/2021-May/060694.html
- https://webcache.googleusercontent.com/search?q=cache:05I4B9PyLz0J:https://wiki.openmoko.org/wiki/Configuring_the_boot_splash_screens&cd=4&hl=de&ct=clnk&gl=de&client=firefox-b-e

ski7777, 2022-12-26:
Lets try it out :)

hypnotoad, 2022-12-27:
fixed in c9c97ce468

closed 2022-12-27 by hypnotoad