ftCommunity/ftcommunity-TXT #213: shut down animation

ski7777, 2022-12-25:
When shutting down the corresponding text is only shown for a fraction of a second and then the screen turn black although the TXT is not shutdown yet. I assume stopping the XServer clears the screen. Is there a way to prevent this behavior?

hypnotoad, 2022-12-25:
I had a quick look to the maemo community but did not find any specific way how they avoided blank screens. One could maybe compare https://github.com/maemo-leste/xserver-xorg-video-fbdev/blob/master/src/fbdev.c to our buildroots fbdev.c.

hypnotoad, 2022-12-25:
I am not very positive that this can be fixed besides waiting for longer until the restart happens.

ski7777, 2022-12-25:
Stupid idea:
Can we take a screenshot with fbgrap right before the XServer is being killed and set it right afterwards via fbset?

hypnotoad, 2022-12-26:
That would be one idea. Another one would be to start the X server as soon as possible during startup (S01) and to stop it as late as possibe. I have a look at that.

hypnotoad, 2022-12-26:
Running the X server with S09 seems to work, but the launcher is still killed too early (andstarting the launcher in S09 or similar will probably lead to side problems).

Probably the best solution would be if the shutdown script (/etc/init.d/rcK) would display some text instead of the UI. Maybe we could use psplash during that as well (with some inverse progress bar or another color).

ski7777, 2022-12-26:
inverse progress bar in red with a text shutting down would be so great

hypnotoad, 2022-12-27:
Fixed in bd8f8d9436

closed 2022-12-27 by hypnotoad