ftCommunity/ftcommunity-TXT #220: radio does not work well (new kernel?)

hypnotoad, 2022-12-27:
The radio app stops working after approx 1 second. There is some spi error. I guess that is related to the minor kernel upgrade: To reproduce:
```
mpg123 -f 8000 -q --stdout --encoding u8 --rate 22050 --mono http://liveradio.sr.de/sr/sr1/mp3/64/stream.mp3 | /home/ftc/apps/ad5e0ea4-4606-4254-add6-60c452c3db36/txt_snd_cat
```

ski7777, 2022-12-28:
I hate kernel and C things...
I will try to debug it :)

ski7777, 2022-12-31:
Just added a package to build the ft_txt_snd_cat in future. I did this to ensure building against matching kernel headers. Sadly still the same behavior

ski7777, 2022-12-31:
Do we want to revert the kernel update?
Does anyone know how to write good C code?
@rkunze @ftrobopy

hypnotoad, 2023-01-01:
Let's try if reverting https://github.com/ftCommunity/ftcommunity-TXT/commit/1503cc8a19cb2386b8c30c92f94d294372ae49fc fixes the problem. I am a little afraid that this bug also affects the ft-server's sound output abilities.

hypnotoad, 2023-01-01:
after reverting  1503cc8, the command above works

ski7777, 2023-05-21:
where did you fix this?

hypnotoad, 2023-05-21:
it was fixed by your revert

ski7777, 2023-05-21:
yes, but we still need to update the kernel. So I want to keep this open until it works with a new kernel