ftCommunity/ftcommunity-TXT #222: make g_ether usable under windows

ski7777, 2022-12-29:
Currently g_ether is not working properly under windows. We should try to be conform to this: https://github.com/fischertechnik/FT-TXT/blob/master/board/FT/TXT/rootfs/etc/init.d/S98usb_g_ether

hypnotoad, 2022-12-29:
Did that work before? Otherwise, I would not assign 1.0 to that.

How can I test that it works?

ski7777, 2022-12-29:
yes and no. The issue is persistent since the beginning. We use the default linux USB vid/pid insted of ft/knobloch+TXT. Workaround: https://forums.raspberrypi.com/viewtopic.php?t=245184#p1496396

hypnotoad, 2022-12-29:
I think we should postpone this for 1.1 in order to avoid delaying the release. I myself never used the txt with a usb connection, so I am not familiar with that (using wifi or serial connection to debug).

closed 2023-02-16 by ski7777