ftCommunity/ftcommunity-TXT #225: Add hashes to all packages

ski7777, 2023-01-02:
- [ ] WARNING: no hash file for ft_bt_remote_server-038b40b431b3d3558b91717d49cd34e1e435dc5a.tar.gz
- [ ] WARNING: no hash file for libroboint-cf25fdba52ad1343317afca683d8b272682f0077.tar.gz
- [ ] WARNING: no hash file for python-ftrobopy-d8eff63eea293f7e3e2d4070e045756fe3c74409.tar.gz
- [ ] WARNING: no hash file for python-fttxpy-0.9.3.tar.gz
- [ ] WARNING: no hash file for python-libroboint-cf25fdba52ad1343317afca683d8b272682f0077.tar.gz
- [ ] WARNING: no hash file for python-robolt-c4800b27669e511f31f85ac3876e1ae0e288c3b3.tar.gz
- [ ] WARNING: no hash file for semantic_version-2.10.0.tar.gz
- [ ] WARNING: no hash file for python-zbarlight-3.0.tar.gz

ski7777, 2023-06-29:
postponing this after the big split-up

hypnotoad, 2025-01-05:
I am closing this issue as there is no progress and my focus is on maintainance currently (https://github.com/ftCommunity/ftcommunity-TXT/discussions/297).

This does not mean that the ticket should not be done, but it is rather to declutter the issue list.

closed 2025-01-05 by hypnotoad