ftCommunity/ftcommunity-TXT #233: Random Kernel Panics when using usb to PC

ski7777, 2023-01-07:
```
[  804.841618] Unhandled fault: external abort on non-linefetch (0x1008) at 0xd0538412
[  804.850156] pgd = c0004000
[  804.853007] [d0538412] *pgd=85e6f811, *pte=47401653, *ppte=47401453
[  804.859634] Internal error: : 1008 [#1] PREEMPT ARM
[  804.864752] Modules linked in: usb_f_ecm g_ether usb_f_rndis u_ether libcomposite configfs wl18xx wlcore mac80211 cfg80211 musb_dsps musb_hdrc wlcore_sdio hci_uart musb_am335x
[  804.881270] CPU: 0 PID: 13 Comm: kworker/0:1 Tainted: G        W       4.14.97 #1
[  804.889109] Hardware name: Generic AM33XX (Flattened Device Tree)
[  804.895530] Workqueue: pm pm_runtime_work
[  804.899741] task: cd920e80 task.stack: cd922000
[  804.904642] PC is at musb_default_readw+0xc/0x1b8 [musb_hdrc]
[  804.910716] LR is at musb_is_tx_fifo_empty+0x38/0x48 [musb_hdrc]
[  804.917011] pc : [<bf028240>]    lr : [<bf0362f8>]    psr: 200c0193
[  804.923576] sp : cd923c98  ip : cd922000  fp : c134c50c
[  804.929050] r10: ccb96010  r9 : 200c0193  r8 : c58524a8
[  804.934526] r7 : c58524d8  r6 : c5821800  r5 : bf02601c  r4 : d0538410
[  804.941364] r3 : d0538412  r2 : 00000003  r1 : 00000002  r0 : d0538410
[  804.948205] Flags: nzCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment none
[  804.955772] Control: 10c5387d  Table: 85804019  DAC: 00000051
[  804.961793] Process kworker/0:1 (pid: 13, stack limit = 0xcd922210)
[  804.968360] Stack: (0xcd923c98 to 0xcd924000)
[  804.972931] 3c80:                                                       c58524a8 c58524a8
[  804.981507] 3ca0: d0538410 bf02601c c5821800 c58524d8 c58524a8 200c0193 ccb96010 bf0362f8
[  804.990084] 3cc0: c585249c c582183c c5821800 bf0378e8 600c0193 c13e1e9e cd922000 c58524a8
[  804.998660] 3ce0: c134c480 c13e1e9e 00000000 bf03788c c138c2bc c019b33c 00000003 c134c500
[  805.007236] 3d00: 6456c6c0 000000bb 6456c6c0 000000bb c13e2c48 c134c480 6456c6c0 00000003
[  805.015812] 3d20: 000000bb 00000000 c134c578 c13e2c5c c13e2c48 c019b930 cd922000 cd922000
[  805.024387] 3d40: c130de00 cd818a64 00000010 cd923df0 00000000 c13e2c5c c13e2c48 c011b4ec
[  805.032962] 3d60: cd922000 c0183660 c1af3998 cd9213f0 c134a640 cd923dac cd818a00 00000002
[  805.041537] 3d80: cd818a64 cd818a00 cd818a64 cd818a64 00000000 cd923df0 cd80c400 cd922000
[  805.050111] 3da0: c134a640 c0183a74 00000001 00000000 cd818a00 cd818a64 00000001 c0186840
[  805.058686] 3dc0: cd818a00 c13dd2dc 00000001 c0182fd0 c0927240 200c0013 ffffffff cd923e24
[  805.067261] 3de0: cd9b18b0 cd922000 cd922000 c0927a70 00000001 ffffff30 00000000 cd920e80
[  805.075836] 3e00: a00c0013 c1323168 c1323168 c011ecc4 cd9b18b0 00000008 cd922000 c134a640
[  805.084411] 3e20: 00000002 cd923e40 c01735e0 c0927240 200c0013 ffffffff 00000051 00000000
[  805.092986] 3e40: 00000000 a00c0013 c1323168 c011dbf8 00000000 00000000 cd9ac100 c011ec6c
[  805.101562] 3e60: cd9b1810 00000000 cd922000 c0599aac 00000000 00000000 c13e2c98 c092720c
[  805.110137] 3e80: cd922000 cd9b1810 04208060 cd922000 0000000a c134a640 00000008 cd922000
[  805.118711] 3ea0: c134a640 c0599ce8 cd9b1810 00000000 c011ecc4 c059a2b4 00000002 00000000
[  805.127285] 3ec0: 00000000 00000001 00000000 c016a8b0 00000000 c0598cbc cd9b1810 cd9b1938
[  805.135861] 3ee0: cd9b18b0 c13421fc cfbc9500 cd923f28 00000001 00000000 c13e2c98 c059ac9c
[  805.144435] 3f00: cd9b1938 cd887500 c13421fc c014f424 00000001 00000000 c014f354 c13421fc
[  805.153010] 3f20: 00000000 c13421fc c1b1f00c c1841908 00000000 c0c1de3c c13421fc cd887500
[  805.161585] 3f40: c13421fc cd887518 c1342230 c134a640 00000008 cd922000 c13421fc c014fd5c
[  805.170160] 3f60: cd920e80 cd887280 cd9180c0 ffffe000 00000000 cd887500 c014fd24 cd8872b8
[  805.178735] 3f80: cd893ec4 c0154ef8 cd922000 cd9180c0 c0154e08 00000000 00000000 00000000
[  805.187308] 3fa0: 00000000 00000000 00000000 c0107bd4 00000000 00000000 00000000 00000000
[  805.195882] 3fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  805.204456] 3fe0: 00000000 00000000 00000000 00000000 00000013 00000000 00000000 00000000
[  805.213139] [<bf028240>] (musb_default_readw [musb_hdrc]) from [<bf0362f8>] (musb_is_tx_fifo_empty+0x38/0x48 [musb_hdrc])
[  805.224721] [<bf0362f8>] (musb_is_tx_fifo_empty [musb_hdrc]) from [<bf0378e8>] (cppi41_recheck_tx_req+0x5c/0xec [musb_hdrc])
[  805.236539] [<bf0378e8>] (cppi41_recheck_tx_req [musb_hdrc]) from [<c019b33c>] (__hrtimer_run_queues.constprop.0+0x1b8/0x5b8)
[  805.248391] [<c019b33c>] (__hrtimer_run_queues.constprop.0) from [<c019b930>] (hrtimer_interrupt+0x80/0x23c)
[  805.258697] [<c019b930>] (hrtimer_interrupt) from [<c011b4ec>] (omap2_gp_timer_interrupt+0x20/0x30)
[  805.268188] [<c011b4ec>] (omap2_gp_timer_interrupt) from [<c0183660>] (__handle_irq_event_percpu+0x6c/0x3d4)
[  805.278489] [<c0183660>] (__handle_irq_event_percpu) from [<c0183a74>] (handle_irq_event+0x48/0xa0)
[  805.287976] [<c0183a74>] (handle_irq_event) from [<c0186840>] (handle_level_irq+0xac/0x1a4)
[  805.296734] [<c0186840>] (handle_level_irq) from [<c0182fd0>] (__handle_domain_irq+0x6c/0xe0)
[  805.305681] [<c0182fd0>] (__handle_domain_irq) from [<c0927a70>] (__irq_svc+0x70/0xb0)
[  805.313986] [<c0927a70>] (__irq_svc) from [<c0927240>] (_raw_spin_unlock_irqrestore+0x70/0x74)
[  805.323018] [<c0927240>] (_raw_spin_unlock_irqrestore) from [<c011dbf8>] (omap_hwmod_idle+0x34/0x44)
[  805.332594] [<c011dbf8>] (omap_hwmod_idle) from [<c011ec6c>] (omap_device_idle+0x3c/0x94)
[  805.341174] [<c011ec6c>] (omap_device_idle) from [<c0599aac>] (__rpm_callback+0x7c/0x268)
[  805.349752] [<c0599aac>] (__rpm_callback) from [<c0599ce8>] (rpm_callback+0x50/0x74)
[  805.357873] [<c0599ce8>] (rpm_callback) from [<c059a2b4>] (rpm_suspend+0x120/0x6d4)
[  805.365902] [<c059a2b4>] (rpm_suspend) from [<c059ac9c>] (pm_runtime_work+0x88/0xa8)
[  805.374027] [<c059ac9c>] (pm_runtime_work) from [<c014f424>] (process_one_work+0x28c/0x7c8)
[  805.382782] [<c014f424>] (process_one_work) from [<c014fd5c>] (worker_thread+0x38/0x508)
[  805.391270] [<c014fd5c>] (worker_thread) from [<c0154ef8>] (kthread+0xf0/0x130)
[  805.398945] [<c0154ef8>] (kthread) from [<c0107bd4>] (ret_from_fork+0x14/0x20)
[  805.406522] Code: ffffe000 e92d47f0 e0803001 e24dd008 (e1d340b0)
[  805.412930] ---[ end trace f13589edbff03b26 ]---
[  805.417775] Kernel panic - not syncing: Fatal exception in interrupt
[  805.424452] ---[ end Kernel panic - not syncing: Fatal exception in interrupt

```

hypnotoad, 2023-01-13:
Is it possible to reproduce this issue deterministically? If yes, it would be great if instructions to reproduce would be added.

hypnotoad, 2023-01-13:
Does this issue happen in 1.0 as well?

ski7777, 2023-01-14:
> Is it possible to reproduce this issue deterministically? If yes, it would be great if instructions to reproduce would be added.

Just plug the USB port of the TXT to a computer and wait... Might take a few minutes 

> Does this issue happen in 1.0 as well?

Yes

closed 2023-02-16 by ski7777