ftCommunity/ftcommunity-TXT #237: update buildroot and decide version rule

ski7777, 2023-01-09:


hypnotoad, 2023-05-20:
I tried the 2023.02 buildroot in https://github.com/ftCommunity/ftcommunity-TXT/compare/br-2023?expand=1, but it does not compile due to some compile problem (afaik it was that QT SIP was updated and does not compile properly).

ski7777, 2023-05-21:
still we need to decide about version rule
Every main release? Only LTS?
Bugfixes?

hypnotoad, 2023-05-21:
2023.02 is actually lts. My own proposal would be to use the most recent lts which works oit of the box.

ski7777, 2023-05-23:
the we should use it with all available bugfixes

hypnotoad, 2023-07-20:
I will give 2023.02.3 a try. That would allow tflite https://pypi.org/project/tflite-runtime/2.13.0/ whic requires some numpy which we don't have yet:
https://git.busybox.net/buildroot/plain/CHANGES?id=2023.02.3

hypnotoad, 2023-07-20:
I fixed the python-sip bug in https://github.com/buildroot/buildroot/pull/131

hypnotoad, 2025-01-05:
Done. 1.1.0 has a quite up-to-date buildroot. 1.1.1 will have the first 2025 release.

closed 2025-01-05 by hypnotoad