ftCommunity/ftcommunity-TXT #248: Rework Organization structure

ski7777, 2023-02-20:
To allow better maintainability I'm currently working on #244 . Next steps would be adding direct support for Raspberry Pi or TXT 4.0.
In my opinion the name `ftcommunity-TXT` is kinda locked on the TXT. We should find a new name!
just `CFW`? or `ftcOS`?
Feel free to suggest something new :)

Additionally we need a new repository structure: `<OSNAME>` will be replaced by the elected OS name
- `<OSNAME>`: Here you can find the central BR2_EXTERNAL tree with all central apps, and a build infratructure to allow building for all platforms
- `<OSNAME>-platform-<PLATFORMNAME>`: Here you can find the platform-specific files (defconfig, dtb, drivers, ...). The defconfig is always named `<PLATFORMNAME>_defconfig`
- `<OSNAME>-app-<APPNAME>`: Just the app files. The app can use GNU Make, CMake, Python, ...
- `<OSNAME>-platform-<PLATFORMNAME>-app-<APPNAME>`: Platform-specific apps

I'm not too sure how to migrate the existing repository, probably we will end up in something like filter-branch. Additionally we should decide on the future use of the ftcommunity-TXT repo. Originally the repo was forked from buildroot, so it contains full buildroot history... In my opinion we should archive this repo and start a whole new tree. Existing issues can be tranfered.

ski7777, 2023-02-20:
Vote here for `CFW`

ski7777, 2023-02-20:
Vote here for `ftcOS

hypnotoad, 2023-02-27:
I would propose to have a new slim core repo which includes your proposals and buildroot as submodules. This new repo should relate different repos with each other (e.g. the UI, the hardware and buildroot).

hypnotoad, 2025-01-05:
I am closing this issue as there is no progress and my focus is on maintainance currently (https://github.com/ftCommunity/ftcommunity-TXT/discussions/297).

This does not mean that the ticket should not be done, but it is rather to declutter the issue list.

closed 2025-01-05 by hypnotoad