ftCommunity/ftcommunity-TXT #284: fischertechnik UI runs as wrong user

hypnotoad, 2023-07-12:
There is a discrepancy between the UID/GID in the rom and the one shipped with FTC. To me, it is unclear which one is considered wrong:
/etc/passwd is
in ROM: `ROBOPro:x:1004:1008::/opt/knobloch:/bin/sh`
in CFW: `ROBOPro:x:1000:1000::/opt/fischertechnik:/bin/sh`

Intuitively, I would change the CFW one, but it has been 1000/1000 from the very first release.
FYI @ski7777  @rkunze  @harbaum

hypnotoad, 2023-07-12:
This leads to the FT-UI being started as user `ROBOPro` of the chroot which actually is the user `sshd` outside the chroot. This user has no write permission to the user folder and thus no programs can be saved.

ski7777, 2023-07-12:
> I'd expect chroot to have the ability to re-map users. After all, changing
> into an alien file system like we do here is the main purpose. Doesn't it?

I just checked on fischertechniks user.tab. As far as I can tell they
generate the user IDs during build and they are not fixed leading to this
problem after changing the user.tab on their side.
I think we can just fix the user ID of ROBOPro to 1008 and it should be
fine.

hypnotoad, 2023-07-12:
run_txtcontrol does a chroot like this:
`chroot /rom/ su - ROBOPro -c "$DISPLAY ./TxtControlMain /dev/ttyO2 65000"`

@harbaum : GNU chroot allows to combine the `su` and the `chroot` with the `--userspec` argument, but BusyBox chroot has no options for that.

The following would maybe work if ROBOPro would be allowed to chroot (but it isn't).
`su - ROBOPro -c " chroot /rom/ su - $DISPLAY ./TxtControlMain /dev/ttyO2 65000"`

closed 2023-07-14 by hypnotoad

hypnotoad, 2023-07-14:
#285 has issues as the ftc user has no access to the sound files. There are 2 options
- give the ftc user access by using some FS in the middle
- do not differentiate between ftc and robopro user - in the robopro chroot it is called robopro, otherwise it is called ftc. The have the same uid:gid 1004:1008

PR #285 (2023-07-14): fix robopro issues
Fix permission and mounting issues that prevent saving and loading robopro files
```
# chroot /rom/ /bin/su - ROBOPro -c "./TxtControlMain /dev/ttyO2 65000"
...
Error writing flash file /opt/knobloch/ROBOPro/test.bin (Permission denied)
...
```

Closes #284