ftCommunity/ftcommunity-TXT #287: ftrobopy has no access to sound files in direct and TA mode

hypnotoad, 2023-07-14:
In direct and TA mode, the ftc user has no access to sound files. That can be reproduced as follows:
```
$ ssh txt
$ python3
Python 3.9.9 (main, Jun 27 2023, 19:38:18) 
[GCC 10.3.1 20210621] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import ftrobopy
>>> txt=ftrobopy.ftrobopy('auto')
Connected to  TXT direct firmware version not detected
>>> txt.play_sound(27, 5)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.9/site-packages/ftrobopy.py", line 3704, in play_sound
  File "/usr/lib/python3.9/site-packages/ftrobopy.py", line 1073, in setSoundIndex
PermissionError: [Errno 13] Permission denied: '/rom/opt/knobloch/SoundFiles/27_Fahrgeraeusch.wav'
```

There are 2 options
- give the ftc user access by using some overlayfs ontop of the rom (there is no ubifs mount option to change permissions)
- do not differentiate between ftc and robopro user - in the robopro chroot it is called robopro, otherwise it is called ftc. The have the same uid:gid 1004:1008

The first option would be more complex. The second would not allow for a simple upgrade path.

harbaum, 2023-07-14:
It may be possible to copy the entire original Rom onto the SD card automatically and then do permissions adjustments and the like there.

hypnotoad, 2023-07-14:
Yes, copying them to /home and bind-mounting this back works.
```
cp -a /rom/opt/knobloch/SoundFiles /home/ROBOPro
chmod 755 /home/ROBOPro/SoundFiles
mount --bind /home/ROBOPro/SoundFiles /rom/opt/knobloch/SoundFiles
```

akki42, 2023-10-07:
Today stumbled over this when trying out Blockly.
@hypnotoad Your work-around helped me out for the time being (i.e. until TXT shutdown/reboot).

However, I tried to fix this "permanently" by adding
`/home/ROBOPro/SoundFiles	/rom/opt/knobloch/SoundFiles	bind	bind		0	0
`
to `/etc/fstab`.
Unfortunately, this is not auto-mounted at startup, but only using `sudo mount -a`.
Any ideas why / how to fix this?
Many thanks!

hypnotoad, 2023-10-08:
I will make this permanent in zhe next release.

hypnotoad, 2023-10-08:
If you would like to fix it Yourself: Put the commands into an init script (the copying and the mounting). PRs are appreciated.

hypnotoad, 2023-10-10:
Should be fixed in 1.0.5: https://github.com/ftCommunity/ftcommunity-TXT/releases/tag/v1.0.5

akki42, 2023-10-13:
Hi @hypnotoad,
Thanks for the quick update. (I had not even tried out the init script solution yet...)

(After undoing my own tweaks,) I installed the update via `sudo system-update 1.0.5` - and everything seems to have worked as planned.
After reboot, `/home/ROBOPro/SoundFiles/` is mounted as `/rom/opt/knobloch/SoundFiles`.

Unfortunately, I can't now re-test my original scenario as Brickly seems to be broken:
In the right hand panel, a spinning wheel is running continously. And whatever block I drag into its main area, the starting block shows an exclamation mark and the run button is greyed / blued out:
[image]

Are there any specific log files that I could look at to see what is wrong? (`/var/log/messages` does not show anything Brickly related, as far as I can see...)
Many thanks!

hypnotoad, 2023-10-13:
Likely something crashed inside brickly. I would bet that it is some minor modification of PyQt namespaces which unfortunately happen from time to time. I think it is easiest to run brickly from the command line via ssh and get the error message there.

akki42, 2023-10-13:
Here is the output:
```
ftc@FT-txt:/> python3 /opt/ftc/apps/user/1f2d90a3-11e9-4a92-955a-73ffaec0fe71/brickly_app.py 
Exctype : <class 'AttributeError'>, value : module 'asyncio' has no attribute 'coroutine' traceback : <traceback object at 0xb43822c8>
Traceback (most recent call last):
  File "/opt/ftc/apps/user/1f2d90a3-11e9-4a92-955a-73ffaec0fe71/brickly_app.py", line 41, in <module>
    class WebsocketServerThread(QThread):
  File "/opt/ftc/apps/user/1f2d90a3-11e9-4a92-955a-73ffaec0fe71/brickly_app.py", line 72, in WebsocketServerThread
    @asyncio.coroutine
     ^^^^^^^^^^^^^^^^^
AttributeError: module 'asyncio' has no attribute 'coroutine'. Did you mean: 'coroutines'?
```

According to [python docs](https://docs.python.org/3.10/library/asyncio-task.html#generator-based-coroutines), "Support for generator-based coroutines is deprecated and is removed in Python 3.11."

As, I believe, this has no longer anything to do with the sound files access, should I open a separate issue?

hypnotoad, 2023-10-14:
should be fixed in the newest version of the brickly app. Updating via the app store should work.

closed 2023-10-14 by hypnotoad