ftCommunity/ftcommunity-TXT #289: robopro programs do not run simtimes in the FT GUI

hypnotoad, 2023-07-23:
it is reported that some rpp programs do not run in offline mode (online is working): https://forum.ftcommunity.de/viewtopic.php?f=33&t=8086&p=62953#p62953.

hypnotoad, 2023-07-23:
Working: [test2.rpp.zip](https://github.com/ftCommunity/ftcommunity-TXT/files/12138303/test2.rpp.zip)

uffi63, 2023-07-23:
Not working: 
[FTS.zip](https://github.com/ftCommunity/ftcommunity-TXT/files/12139028/FTS.zip)

hypnotoad, 2023-07-25:
I guess this will be a difficult problem. I could reproduce the problem with the following commands:
```
ssh txt
su -
pgrep xinit | xargs kill
chroot /rom/ /bin/su - ROBOPro -c "strace -f -e open ./TxtControlMain /dev/ttyO2 65000"
```
To stop the TxtControlMain, one can use another terminal:
```
su -
pgrep -u ROBOPro | xargs kill -9
```

When the file is loaded from flash, there is a segmentation fault. The strace log indicates it is sent to a thread which does something with the sound system (1539 in [nok.txt](https://github.com/ftCommunity/ftcommunity-TXT/files/12165549/nok.txt and 1573 in [ok.txt](https://github.com/ftCommunity/ftcommunity-TXT/files/12165551/ok.txt). Permissions of /dev/video0 and the /dev/snd files seem all ok.

hypnotoad, 2023-07-25:
test2.rpp (which did not crash) does not mention video initialization: [test2.txt](https://github.com/ftCommunity/ftcommunity-TXT/files/12165601/test2.txt)

hypnotoad, 2023-07-25:
I would continue to test which component causes the behaviour. When I add a camera to test2.rpp, it still works
[test3.rpp.zip](https://github.com/ftCommunity/ftcommunity-TXT/files/12165702/test3.rpp.zip). I don't see anything special in the log files:
[test3.txt](https://github.com/ftCommunity/ftcommunity-TXT/files/12165718/test3.txt)