ftCommunity/ftcommunity-TXT #292: Date / time not updated at startup

akki42, 2023-10-14:
Using the newly released [v 1.0.5](https://github.com/ftCommunity/ftcommunity-TXT/releases/tag/v1.0.5), my TXT does not update date & time at startup.

According to `/var/log/messages`, `ntpd` is supposed to run as user `ntp`, which is not configured.
```
Jan  1 01:00:15 FT-txt daemon.notice ntpd[1067]: ntpd 4.2.8p15@1.3728-o Do 20. Jul 10:39:46 UTC 2023 (1): Starting
Jan  1 01:00:15 FT-txt daemon.notice ntpd[1067]: Command line: /usr/sbin/ntpd -u ntp:ntp -g -p /var/run/ntpd.pid
Jan  1 01:00:15 FT-txt daemon.notice ntpd[1067]: ----------------------------------------------------
Jan  1 01:00:15 FT-txt daemon.notice ntpd[1067]: ntp-4 is maintained by Network Time Foundation,
Jan  1 01:00:15 FT-txt daemon.notice ntpd[1067]: Inc. (NTF), a non-profit 501(c)(3) public-benefit
Jan  1 01:00:15 FT-txt daemon.notice ntpd[1067]: corporation.  Support and training for ntp-4 are
Jan  1 01:00:15 FT-txt daemon.notice ntpd[1067]: available at https://www.nwtime.org/support
Jan  1 01:00:15 FT-txt daemon.notice ntpd[1067]: ----------------------------------------------------
Jan  1 01:00:15 FT-txt daemon.info ntpd[1075]: proto: precision = 5.875 usec (-17)
Jan  1 01:00:15 FT-txt daemon.info ntpd[1075]: basedate set to 2023-07-08
Jan  1 01:00:15 FT-txt daemon.info ntpd[1075]: gps base set to 2023-07-09 (week 2270)
Jan  1 01:00:15 FT-txt daemon.info ntpd[1075]: Listen and drop on 0 v6wildcard [::]:123
Jan  1 01:00:15 FT-txt daemon.info ntpd[1075]: Listen and drop on 1 v4wildcard 0.0.0.0:123
Jan  1 01:00:15 FT-txt daemon.info ntpd[1075]: Listen normally on 2 lo 127.0.0.1:123
Jan  1 01:00:15 FT-txt daemon.info ntpd[1075]: Listen normally on 3 lo [::1]:123
Jan  1 01:00:15 FT-txt user.info kernel: [   22.313274] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
Jan  1 01:00:15 FT-txt daemon.info ntpd[1075]: Listening on routing socket on fd #20 for interface updates
Jan  1 01:00:15 FT-txt daemon.info ntpd[1075]: kernel reports TIME_ERROR: 0x41: Clock Unsynchronized
Jan  1 01:00:15 FT-txt daemon.info ntpd[1075]: kernel reports TIME_ERROR: 0x41: Clock Unsynchronized
Jan  1 01:00:15 FT-txt daemon.err ntpd[1075]: Cannot find user `ntp'
```

Manually calling ntpd without the -u option works fine:
```Jan  1 01:01:40 FT-txt authpriv.info sudo: pam_unix(sudo:session): session opened for user root(uid=0) by ftc(uid=1001)
Jan  1 01:01:40 FT-txt daemon.notice ntpd[1149]: ntpd 4.2.8p15@1.3728-o Do 20. Jul 10:39:46 UTC 2023 (1): Starting
Jan  1 01:01:40 FT-txt daemon.notice ntpd[1149]: Command line: /usr/sbin/ntpd -d -g -p /var/run/ntpd.pid
Jan  1 01:01:40 FT-txt daemon.notice ntpd[1149]: ----------------------------------------------------
Jan  1 01:01:40 FT-txt daemon.notice ntpd[1149]: ntp-4 is maintained by Network Time Foundation,
Jan  1 01:01:40 FT-txt daemon.notice ntpd[1149]: Inc. (NTF), a non-profit 501(c)(3) public-benefit
Jan  1 01:01:40 FT-txt daemon.notice ntpd[1149]: corporation.  Support and training for ntp-4 are
Jan  1 01:01:40 FT-txt daemon.notice ntpd[1149]: available at https://www.nwtime.org/support
Jan  1 01:01:40 FT-txt daemon.notice ntpd[1149]: ----------------------------------------------------
Jan  1 01:01:40 FT-txt daemon.info ntpd[1151]: proto: precision = 5.875 usec (-17)
Jan  1 01:01:40 FT-txt daemon.info ntpd[1151]: basedate set to 2023-07-08
Jan  1 01:01:40 FT-txt daemon.info ntpd[1151]: gps base set to 2023-07-09 (week 2270)
Jan  1 01:01:40 FT-txt daemon.info ntpd[1151]: Listen and drop on 0 v6wildcard [::]:123
Jan  1 01:01:40 FT-txt daemon.info ntpd[1151]: Listen and drop on 1 v4wildcard 0.0.0.0:123
Jan  1 01:01:40 FT-txt daemon.info ntpd[1151]: Listen normally on 2 lo 127.0.0.1:123
Jan  1 01:01:40 FT-txt daemon.info ntpd[1151]: Listen normally on 3 wlan0 192.168.2.15:123
Jan  1 01:01:40 FT-txt daemon.info ntpd[1151]: Listen normally on 4 lo [::1]:123
Jan  1 01:01:40 FT-txt daemon.info ntpd[1151]: Listen normally on 5 wlan0 [fe80::cadf:84ff:fe44:921c%3]:123
Jan  1 01:01:40 FT-txt daemon.info ntpd[1151]: Listening on routing socket on fd #22 for interface updates
Jan  1 01:01:40 FT-txt daemon.info ntpd[1151]: kernel reports TIME_ERROR: 0x41: Clock Unsynchronized
Jan  1 01:01:40 FT-txt daemon.info ntpd[1151]: kernel reports TIME_ERROR: 0x41: Clock Unsynchronized
Jan  1 01:01:40 FT-txt authpriv.info sudo: pam_unix(sudo:session): session closed for user root
Oct 14 18:05:49 FT-txt daemon.info ntpd[1151]: receive: Unexpected origin timestamp 0xbc17c26b.ec5cc3cf does not match aorg 0000000000.00000000 from server@23.88.7.139 xmt 0xe8d53add.608bf3f0
```

(Without automatic or manual time setting, there are some resulting errors, e.g. Store app returns "Netzwerkfehler: SSL-Aushandlung fehlgeschlagen"...)

hypnotoad, 2023-10-15:
I cannot immediately reproduce the problem. I see two differences in my setup:
1. in /var/log/messages, the time is correct from the first line
2. I do not have this message about ntp

I guess that that some update did not work smoothly. My setup is not clean either. Before re-installing from scratch: Could you check if ntp is part of your /etc/passwd? I have
```
# cat passwd | grep ntp
ntp:x:101:102:ntpd user:/:/bin/false
```

hypnotoad, 2023-10-15:
My guess is that in your case, the time battery is empty as well!

akki42, 2023-10-15:
Time battery indeed is (or seems to be) empty. I'll get a replacement asap.

Regarding user ntp: no such user is present in my system.

Full /etc/passwd reads as follows:
```
ftc@FT-txt:~> cat /etc/passwd
root:x:0:0:root:/root:/bin/sh
daemon:x:1:1:daemon:/usr/sbin:/bin/false
bin:x:2:2:bin:/bin:/bin/false
sys:x:3:3:sys:/dev:/bin/false
sync:x:4:100:sync:/bin:/bin/sync
mail:x:8:8:mail:/var/spool/mail:/bin/false
www-data:x:33:33:www-data:/var/www:/bin/false
operator:x:37:37:Operator:/var:/bin/false
nobody:x:65534:65534:nobody:/home:/bin/false
ROBOPro:x:1004:1008::/opt/fischertechnik:/bin/sh
ftc:x:1001:1001::/home/ftc:/bin/sh
named:x:1000:1000:BIND daemon:/etc/bind:/bin/false
dbus:x:1002:1002:DBus messagebus user:/run/dbus:/bin/false
sshd:x:1003:1003:SSH drop priv user:/var/empty:/bin/false
ftc@FT-txt:~> 
```

This is a rather fresh installation: newly installed v. 1.0.4 on October 7th and upgraded to v. 1.0.5 via `sudo system-update 1.0.5` this Friday.

I can of course test a fresh v. 1.0.5 installation. The only reason I haven't done so already is that my wifi password contains a character (~) not shown in the TXT's wifi settings, so getting the TXT onto the network is a bit of a hassle.

hypnotoad, 2023-10-15:
The problem happens during the upgrade of 104 to 105 and it can be fixed manually with
```
sudo adduser -s /bin/false -S -D -H -u 101 -G ntp ntp
```

I added this to the upgrade instructions. The tilde Logo will be on the keyboard in the next release.

closed 2023-10-15 by hypnotoad